PX4/PX4-Flow #32: Sonar Performance Degradation

jgoppert, 2015-01-20:
Master 91e6
[image]

Ae201 (before mode filter etc.)
[image]

I believe that the complexities introduced by the on-board filtering are unwarranted for the performance gain. It also is causing a loss of information as the 0 return code is mapping both out of range close and out of range far to the same value.  As a possible solution, I created a branch that adds an error status to the i2c frame (nominal, unint, low, high): https://github.com/jgoppert/Flow/tree/sonar-status  This would also require the same change on the FMU side and some modification to position_estimator_inav.

LorenzMeier, 2015-01-21:
The suggestion makes sense.

LorenzMeier, 2015-04-24:
Hi James, how is this PR looking? Any chance to take another stab getting this into master? The issue I'm having is that the flow driver will probably silently fail - we need to introduce a version tag and escalate the issue to the GCS. Or we need to map near and far to e.g. -1 and -2 and keep the frame format intact.