PX4/PX4-Flow #44: Error in camera sensor initialisation

xpetresx, 2015-02-16:
Hello,
I would like to report a minor error in MT9V034 camera sensor initialization for PX4Flow module which I have found. 
The error is in the file mt9v034.c[0] where there is on line 183 (197 for Context B respectively)
```
mt9v034_WriteReg16(MTV_V2_CTRL_REG_A, total_shutter_width);
```
where MTV_V2_CTRL_REG_A is 0x32 and total_shutter_width is defined as 0x1e0 (480 Dec).

I suppose that this is incorrect because according to the datasheet[1] V2_CTRL_REG_A is a voltage control register with allowed values from 0 to 63 and default value of 0x1e used for HDR knee thresholds. I suppose that there should be
```
183   mt9v034_WriteReg16(MTV_COARSE_SW_TOTAL_REG_A, total_shutter_width);
197   mt9v034_WriteReg16(MTV_COARSE_SW_TOTAL_REG_B, total_shutter_width);
```
respectively.
Addressing issue of DrTon #41 on a garbage image this is caused by the "feature" of the sensor. It simply stops working when there is massive change in lighting conditions. I suppose that it is somehow related with the AEC and AGC features of the sensor. It is also greatly influenced by how well the lens is focused. I am currently performing experiments with an FPGA based module with mt9v034 sensor so I will post the solution if I will resolve the problem.

Have a nice day,
Petr

[0] https://github.com/PX4/Flow/blob/master/src/mt9v034.c
[1] http://www.aptina.com/assets/downloadDocument.do?id=406

liubenyuan, 2015-02-17:
Great ! I am also working on a FPGA (Xilinx Zynq) based mt9v034 solution :)

LorenzMeier, 2015-12-10:
Fixed in 04b6932, thanks!

commit 04b69322 (2015-12-10): Fix shutter width, fixes #44
closed 2015-12-10 by LorenzMeier