PX4/PX4-Flow #62: README.md outdated, upload-usb no long works

jlian, 2015-09-04:
To build,  `make archives` should be ran before `make`.

Also, `make upload-usb` results in `make: *** No rule to make target 'upload-usb'.  Stop.` Thus, I have not found a way to flash the unit using the command line. I have tried to run `make px4flow-v2_default upload` but that results in
```
make[2]: ~/PX4/Flow/makefiles//baremetal/upload.mk: No such file or directory
make[2]: *** No rule to make target `~/PX4/Flow/makefiles//baremetal/upload.mk'.  Stop.
make[1]: *** [upload] Error 2
```
Motivation: I am currently trying to flash the PX4Flow unit without the use of qGroundControl. In particular, I'm trying to flash [this .px4 file](http://ardupilot.com/downloads/?did=118) on it. It looks like the current release of qGroundControl (v2.7.1) does not allow flashing with custom firmware on the PX4Flow. so I'm trying to do it via the command line.

LorenzMeier, 2015-09-04:
It does support custom firmware.

jlian, 2015-09-04:
With qGroundControl 2.7.1 and PX4Flow? This is the output I get when I tried to flash the firmware with qGroundControl
```
QGroundControl can upgrade the firmware on Pixhawk devices, 3DR Radios and PX4 Flow Smart Cameras.
Plug in your device via USB to start firmware upgrade
Found device: PX4 Flow
Connected to bootloader:
Version: 4
Board ID: 6
Flash size: 1032192
Downloading firmware...
From: http://px4-travis.s3.amazonaws.com/Flow/master/px4flow.px4
Download complete
Firmware file missing parameter_xml_size key
Firmware file missing airframe_xml_size key
Succesfully decompressed image
Erasing previous program...
Erase complete
Programming new version...
Program complete
Verifying program...
Verify complete
Upgrade complete
------------------------------------------
```
I could not find a way to get it to not download and flash firmware from the internet. What am I missing?

Thanks.

jlian, 2015-09-04:
I hope I'm not missing something really obvious. Here is a quick video showing what it looks like on my system during the firmware flashing process with QGroundControl 2.7.1: https://youtu.be/NRMZcw-8fQk

LorenzMeier, 2015-09-04:
@davids5 Could you help to fix the boot loading in the repo? Please push straight to master as needed.

LorenzMeier, 2015-09-04:
You are right - we'll fix this and David will provide a fixed upload script.

davids5, 2015-09-04:
@LorenzMeier, @jlian 

PR  https://github.com/PX4/Flow/pull/56 is in - I do not have write access

davids5, 2015-09-05:
@jlian Please give the latest master a go.

jlian, 2015-09-05:
Thanks David, looks like it works now!

jlian, 2015-09-05:
Actually, running `make px4flow-v2_default upload` still gives errors: 
```
make[2]: ~/PX4/Flow/makefiles//baremetal/upload.mk: No such file or directory
make[2]: *** No rule to make target `~/PX4/Flow/makefiles//baremetal/upload.mk'.  Stop.
make[1]: *** [upload] Error 2
```
Am I doing something wrong still? And just to confirm, `make upload-usb` works now.

davids5, 2015-09-05:
make  upload-usb  was the legacy way so I kept it.  I left the "make upload" as is (the help suggested it is not implemented on all platforms)

```make px4flow-v2_default upload-usb``` should work.

Do you have V2 HW? Please If not use only px4flow-v1_default.

jlian, 2015-09-05:
Yeah sorry about that, I edited my previous comment. Everything is good now, thanks guys!

closed 2015-12-10 by LorenzMeier