PX4/PX4-Flow #63: something was wrong  when compiled

ckatgit, 2015-09-09:
%% Generating UAVCAN Bootable \org.pixhawk.px4flow-v2\-2.0-0.1.3ddba48f as /c/Users/ASUS/Flow/Images/\org.pixhawk.px4flow-v2\-2.0-0.1.3ddba48f.uavcan.bin
BIN:     /c/Users/ASUS/Flow/Build/px4flow-v2_default.build/firmware.bin
Traceback (most recent call last):
  File "c:/Users/ASUS/Flow/Tools/make_can_boot_descriptor.py", line 278, in <module>
    with FirmwareImage(in_file, "rb") as in_image:
  File "c:/Users/ASUS/Flow/Tools/make_can_boot_descriptor.py", line 86, in __init__
    self._file = open(path_or_file, mode + "b")
ValueError: Invalid mode ('rbb')
make[1]: Leaving directory `/c/Users/ASUS/Flow/Build/px4flow-v2_default.build'
make[1]: *** [/c/Users/ASUS/Flow/Build/px4flow-v2_default.build/firmware.bin] Error 1
make: *** [/c/Users/ASUS/Flow/Build/px4flow-v2_default.build/firmware.px4] Error 2

molyswu, 2015-10-21:
my project build for make px4flow-v2_default failed

LorenzMeier, 2015-10-21:
That hardware is not yet available anyway, so not an issue.

closed 2015-10-21 by LorenzMeier