PX4/PX4-Flow #77: what's the KLT method reference

xiaozhuka, 2016-05-06:
Hi all, can any one tell me what's the KLT method reference

LorenzMeier, 2016-05-06:
http://lmgtfy.com/?q=KLT+optical+flow

closed 2016-05-06 by LorenzMeier

tpetri, 2016-05-07:
Sorry for posting into closed issue, but I thought about a reference because the original papers might not be that efficient to understand and map to the code. I think a good point to start is "Lucas-Kanade 20 Years On: A Unifying Framework" (https://www.cs.cmu.edu/afs/cs/academic/class/15385-s12/www/lec_slides/Baker&Matthews.pdf).
For the case of PX4Flow just assume that the warp is simply the x and y displacement estimated with the inverse compositional method (by using Gauss-Newton), which is essentially the original method without the affine warp estimation and formulated in a slightly more efficient manner.