PX4/PX4-Flow #79: MT9v034 Camera

doctorluhang, 2016-06-06:
Dear all,
I have a simple question. Can I change the camera sensor to anthoer one? Since this sensor mt9v034 requires a lens that is pretty large, I want to change it into another smaller camera such as OVA7670, etc. Can I do that?

LorenzMeier, 2016-06-06:
As long as its global shutter yes.

doctorluhang, 2016-06-06:
@LorenzMeier Thanks for your quick reply. Could you please tell me why it requires global shutter? IMHO, it only captures the video frames that seemingly does not need to shutter. Besides, MT9v034 has pins of exposure and standby. Are these functions used in calculating in optical flow?

tpetri, 2016-06-08:
He was talking about the electrical shutter of the image sensor. If you don't want to have a global shutter, you most likely mean a rolling shutter sensor, then I would suggest to take the [rolling shutter effect](https://en.wikipedia.org/wiki/Rolling_shutter) into consideration. This is something that you don't want to have because the existing code assumes that there are no rolling shutter effects in the images and it would lead to potentially very distorted estimations.

closed 2020-02-14 by mike239x