PX4/PX4-Flow #80: What is the resolution that STM32F427 handles?

doctorluhang, 2016-06-07:
As we know, the SRAM in F427 is only 256KB, so if it captures the full 640x480 resolution image, the sram cannot store it. Does this pix4flow configures the camera sensor at a lower resolution, say 320x240 or something else?

tpetri, 2016-06-07:
if you start looking at the code, lets say main.c, you'll quickly find this line:
https://github.com/PX4/Flow/blob/4a314cfdb099aed9795b825e7518203771207fbb/src/modules/flow/main.c#L89

A fulltext search for FULL_IMAGE_SIZE in this repo (top search bar) gives this result:
https://github.com/PX4/Flow/blob/4a314cfdb099aed9795b825e7518203771207fbb/src/include/mt9v034.h#L127

doctorluhang, 2016-06-08:
@tpetri Thanks for the reply. Then I want to ask if this 188x120 resolution even enough for calculating optical flow. Why does the FLOW use this resolution? Is there any intrinsic reasons?

tpetri, 2016-06-08:
I don't know what you exactly mean with not enough resolution? I saw the sensor working, at least it looked like it was successfully computing optical flow, many others would also claim the sensor can work.
Did you check how the flow is computed (flow.c)? Its SAD correlation (the old algorithm) or SAD minimization (klt version) of 8x8 pixel patches at a very small number of sample locations. Its less about the sensor resolution than what is visible on the image.
One reason is mainly the limited processing power and the trade-off for high frequency processing. If you can provide an implementation that can deal with more resolution providing more accuracy and equal robustness, I think everyone is happy to get a merge request, the implementation can for sure be further improved.

closed 2020-02-14 by mike239x