PX4/PX4-Flow #81: Gyro data question

doctorluhang, 2016-07-02:
Dear experts,
I have a question regarding the gyro data manipulation in the main.c line 407:

		/* gyroscope coordinate transformation */
		float x_rate = y_rate_sensor; // change x and y rates
		float y_rate = - x_rate_sensor;
		float z_rate = z_rate_sensor; // z is correct

I do not understand why x and y data is assign in reverse. I thought that even if the gyro is in up down direction when using the board but y direction also points to the front, so why does y data is assigned to x_rate? please clarify me. Thanks!

mike239x, 2020-01-30:
I should say in advance I probably don't understand what I'm saying, so be careful and think for yourself.

I think the reason is the following:

What is being computed here is offset to the flow based on gyro data.

Assuming the camera is stationary, if we rotate it along X axis, the image from camera would move along Y axis, as if the camera moved along Y axis horizontally. This is exactly why X-s and Y-s get swapped like that. Similar effect for rotating along Y axis - it results in the camera image looking as if it moves along the X axis. The minus sign appears because... well... it's hard to explain without a drawing...

Side note: rotation along Z axis would not move the image at all, and instead would rotate the image, so I think that one will be treated completely differently in the code.

mike239x, 2020-01-30:
Also, according to https://github.com/szobov/Flow/commit/7aa35163514cf908f5cb90bfb56cf69623906524
the signs are wrong, I guess.

tpetri, 2020-02-14:
The cited code in main.c might just be a hardware transform (maybe the sensor is physically rotated against the printed coordinate axis on the px4flow pcb). So that snippet makes sure that x_rate and y_rate correspond with the printed axis.

If that is the case then the sign switch in the commit of szobov might not be correct. Because we need exactly the transform that mike239x explained to match the rates with the flow values. It might be confusing to see the same transformation twice.

To verify that we would need to know the actual coordinate axis of the gyro sensor. It could be that during the design of the pcb the sensor was mounted such that the rate readings of the sensor could be directly used for compensation.