PX4/PX4-Flow #82: Position value from PX4Flow is random

harewei, 2016-07-20:
I've been trying to read the X and Y values from PX4Flow for quite a while now (and trying to use offboard indoor control with it).

I've been able to get flow_x and flow_y using mavlink message, but even if I place the PX4Flow in a stationary position, these two values seem completely random and jumps around a lot.  Is this suppose to be correct?  Maybe the Kalman Filter or some other functions in PX4 firmware is able to decode these seemingly random values somehow?  If so, could someone tell me how to get the drone's XY position based on the signal from PX4Flow?

Thanks.

jgoppert, 2016-07-20:
In qgroundcontrol if you plug in the px4flow you can select position publication and it gives you what you want, this just adds the flow on the px4flow instead of the px4 fmu.

jgoppert, 2016-07-20:
Also see the tutorial on dev.px4.io on optical flow.

harewei, 2016-07-20:
Um could you be a bit more specific where I can find the "position publication"?
I have QGroundControl v2.9.4-218-g7c6397d but can not find this anywhere.  There is a system parameter called "Position", but if I turn that from 0 to 1, then PX4Flow video turns to black.  Btw I'm checking the mavlink messages using QGroundControl Widgets -> Mavlink Inspector

jgoppert, 2016-10-09:
Sys send lpos

ChristophTobler, 2017-08-16:
@harewei Please reopen if still an issue

closed 2017-08-16 by ChristophTobler