PX4/PX4-Flow #83: SF10 Lidar replacing HRLV Sonar in Px4flow

bhaskarjyotidutta, 2016-08-05:
Recently I replaced the HRLV MaxBotics Sonar of PX4Flow with SF10 Lidar . SF10 Lidar's range is 0 to 50 meters as compared to  HRLV's 0 to 5 meter.

I managed to integrate the SF10 Lidar successfully but the values won't go beyond 9.9 meter although I made all the necessary changes in the Sonar.c code 

Is there any hardware or library limitation in PX4Flow that the range of any range finder integrated to it won't go beyond 9.9 meters.

Itamare4, 2019-04-28:
Hi @bhaskarjyotidutta,
Could you please share the code for integrating SF10 at sonar.c?

Thanks.

zaccer, 2020-01-02:
HI @bhaskarjyotidutta, how did you connect the LiDAR to px4flow module and what changes did you make in the firmware to read LiDAR data instead of sonar?
I am trying to do the same thing with tfmini.