PX4/PX4-Flow #84: Understanding the flow code (flow.c)

IkerZubi, 2016-09-09:
Hi!

I have actually bought a PX4FLOW. It is the first time I work with image processing and I am really interested of knowing about how it works. I have been trying to understand the code but I have some questions. 
My problem is not the sintaxis of the code (I can find it on the internet), what I don't understand is the use/ utility of some functions. I think that with a short definition of some functions should be enough to start understanding everything.

I know that it is based on SAD algorithm which takes the sum of the absolute diferences on each pixel. I have seen that the lower is the sum, the more similar are the 2 images. The functions I don't understand are the next ones:
 
1.-
https://github.com/PX4/Flow/blob/4a314cfdb099aed9795b825e7518203771207fbb/src/modules/flow/flow.c#L62
I think it is doing the SAD algorithm taking the difference on the images?

2.-
https://github.com/PX4/Flow/blob/4a314cfdb099aed9795b825e7518203771207fbb/src/modules/flow/flow.c#L62
Why it is a 4x4 pattern taken?
I have been reading about why is interesting to take the average pixel gradient but I continue without understanding why is so interesting and why it is used for knowing if the pixel is suitable for flow tracking:
https://github.com/PX4/Flow/blob/4a314cfdb099aed9795b825e7518203771207fbb/src/modules/flow/flow.c#L433 

3.-https://github.com/PX4/Flow/blob/4a314cfdb099aed9795b825e7518203771207fbb/src/modules/flow/flow.c#L221

I think I understand this function : It does the SAD algorithm for a 8x8 pattern taken from the two photos of (188*120). It only takes a 8x8 pattern to make the process quicklier and computable, isn't it?. For analyzing the pixels, it analyze each bit of each pixel and takes de diference with the other image, so the ABSDIFF of EACH BIT of EACH PIXEL should be 1 or 0? It takes a lot of different 8x8 patterns of each photo?

4.- https://github.com/PX4/Flow/blob/4a314cfdb099aed9795b825e7518203771207fbb/src/modules/flow/flow.c#L351

I can see the difference from the above function. Isn't the SAD computation done in the previous step function? Why it is doing again the SAD compute?

That's all, I think if I understand what this functions are used for (Not how are done, why are used) I will be able to understand everything. I have surfing on the internet and I think I understand SAD algorithm, but I don know how it is implemented.

Thank you for your help :)

mike239x, 2020-02-05:
Regarding 4. - as you can see from https://github.com/PX4/Flow/blob/master/src/modules/flow/flow.c#L452, `compute_sad_8x8` is currently unused and `ABSDIFF` macro is used instead.
Regarding 8x8 pixel patches used - indeed, the code has a lot of intrinsics to make it fast, which enforces "fixed" size.
Can't really answer any other questions you asked.

mike239x, 2020-02-06:
So yeah, `compute_sad_8x8` and `ABSDIFF` have the same functionality, but the macro "turned out to be 25% faster". The macro however has hard-coded values for image width, so I wonder if it is a fair comparison.

Reason for 4x4 pattern is probably "just because".  STM32 has intrinsics that work with 4 bytes, so the choice was either 4x4 or 8x8. I don't know why authors ended up with 4x4. The only really important part about image patch is that it has at least some sort of nonhomogeneity to it, that is, it is not just one and the same color all over the place.

Regarding 3:  "It only takes a 8x8 pattern to make the process quicklier and computable, isn't it?" - yes; 
"the ABSDIFF of EACH BIT of EACH PIXEL should be 1 or 0?" - no no no. A single pixel brightness value is encoded with one BYTE (not bit), and `ABSDIFF` also calculates SAD on a byte-wise level on a patch of 8x8 pixels (8 rows by 8 bytes).

closed 2020-02-14 by mike239x