PX4/PX4-Flow #85: Increase the maximum exposure

jgoppert, 2016-09-17:
The maximum exposure is now set at 5% of the default. This is really detrimental indoors and for low light settings outdoors. I have manually set it to 50% of the default and it is much more robust.

@LorenzMeier

LorenzMeier, 2016-09-17:
I think there were concerns about motion blur. If your testing proves that that's actually not an issue we should go ahead and change the default.

jgoppert, 2016-09-18:
Yeah at default exposure there is a lot of motion blur, optimal value is
somewhere around 25-50% of default from testing indoors. Will try to get
some logs to compare.

Oceasdeep, 2018-08-06:
I used px4flow with all default values, px4flow must be installed more than 30 cm above the ground, in order to take off on its own.I changed the lens of 6mm and changed the focal length parameter to 6. Can be installed normally without raising the height, but use 6mm lens in a slightly brighter lighting environment, the drone positioning effect has drifted, sometimes drifting very large.have I fogot set some parameters？how i solve this problem?waiting for your reply,thanks in advance.
@LorenzMeier