PX4/PX4-Flow #87: Very slow I2C update rate

thomasck, 2016-11-28:
The flow measurements I am getting via I2C are only available at around 15 Hz, whereas I believe they should be available at about 400 Hz. Is there something I am supposed to change after I download the firmware?

Thanks!

Thomas

jgoppert, 2016-11-29:
Can you give more details on your setup and how you are measuring the i2c update rate?

LorenzMeier, 2016-11-29:
The optical flow is running at 400 Hz but you are getting it pre-integrated at a lower rate. There is no benefit in running the interface faster and this is not an actual issue with the sensor. You can change the interface code if you want to read it faster, but it will make no difference in terms of the result.

closed 2016-11-29 by LorenzMeier

thomasck, 2016-11-29:
What do you mean by pre-integrated? I am trying to get a current body frame x/y velocity measurement at each time step in my control loop, which is running between 80-100 Hz. But the px4flow data seems to be updating way slower than that so I get lot's of repeated values then sudden jumps and then more repeated values, which results in an overall very blockish signal that will not work as feedback. I'm begging to suspect it might be how I am reading the registers.. is there any protocol to reading or resetting the i2c _frame registers? I am using linux's mraa i2c class and reading individual words from the i2c_frame.

jgoppert, 2016-11-29:
The bandwidth for position control is only ~5 Hz. So getting the data at 10 Hz is all you need. When each camera image is taken and flow is computed, it stores the pixel flow and adds it up over the interval of 0.1 seconds. If you use my latest pull request, you can change the output rate if you like in qgroundcontrol, but will have to change it in the firmware to save it. https://github.com/PX4/Flow/pull/86/files#diff-7624ab7c53eb9962990d9668d74d8608R198

Your velocity estimation should use a kalman filter and discretely update when it gets a new flow message only. If you keep updating using old flow data it will create a blocky signal. Look at the px4/firmware driver for the flow to compare how you are checking if the data is new or not.

thomasck, 2016-11-29:
Hmm interesting, what about the bandwidth for velocity control? That is what I'm really trying to do. What measurement model do I use for the kaman filter?

jgoppert, 2016-11-29:
5 hz should be fine, if you have an imu you should use the specific acceleration to predict the velocity: d/dt(v) = (a - g), and then the measurement is linear in the state, y = v

thomasck, 2016-12-12:
I am a little confused about how to implement this - doesn't d/dt(v) = (a - g) have to be observable?

thomasck, 2017-01-06:
Thanks for your help @jgoppert, I was able to implement the filter and get better velocity data. However I'm still trying to change the sonar update rate to around 100 hz. Is this just a matter of changing a parameter somewhere?  I downloaded your updated firmware from google drive, but am unsure how to load it into my px4 module? QGroundControl won't let me load it as custom firmware.

Thanks,

Thomas