PX4/PX4-Flow #89: Cannot read values from PX4Flow

kanishkanamdeo, 2017-04-15:
So I have bought this PX4Flow clone from Aliexpress and it's not working. The only output I get is the video feed that I can view on the QGC. All other values are not returned. Help?

trigger1996, 2017-04-24:
Lots of guys said that optflow didn't work well.
Look up these in ardupilot or pixhawk official site, and, don't forget to make some changes in flow
and I've tried, seems ardupilot 3.5.0-dev doesn't support optflow, perhaps the earlier version works.

https://dev.px4.io/en/tutorials/optical_flow.html
and sorry I haven't found any ardupilot documents in English

ChristophTobler, 2017-08-16:
I'm sorry, but I guess we can't help you with hardware clones as we don't use/have them.

closed 2017-08-16 by ChristophTobler