PX4/PX4-Flow #95: can not connect to QGC

feilongfl, 2017-08-18:
i use the command below to upload firmware.and then i can not connect to the QGC,what should i do.
```
make px4flow-v1_default -j4 -l4 upload-usb
```
output:
[image]

QGC:
[image]

code is in master branch.

feilongfl, 2017-08-18:
i have join the user group dialout

feilongfl, 2017-08-18:
then i try to connect with mavproxy,it show me `Device or resource busy`
[image]

ChristophTobler, 2017-08-18:
There is an error: `ERROR: timeout waiting for data`
 
It should be
```
Found board 6,0 bootloader rev 3 on /dev/ttyACM0
erase...
program...
verify...
done, rebooting.
```
Try again with `make upload-usb`

feilongfl, 2017-08-18:
oh sorry, i put the wrong picture on it,it not the program downloads error.

i think it may be the QGC 's error.
i have try to use old QGC,it works,now i use the QGC version v3.1.1

closed 2017-08-18 by ChristophTobler

feilongfl, 2017-09-01:
i put an issues in [https://github.com/mavlink/qgroundcontrol/issues/5591](issues),they reply me it will be fix in v3.3