PX4/PX4-Flow #98: There might be a hidden error in hist computing

leonkaihao, 2017-12-21:
The source line can be found in function compute_flow as follow: 
`if (subdirs[i] == 0 || subdirs[i] == 1 || subdirs[i] == 7) hist_index_x += 1;`
It is strange that the variable i has no direct relation with the array ‘subdirs‘， so there must be some misunderstanding with this array, or using 'meancount-1' as its index.

tpetri, 2017-12-21:
I agree, this seems to be a quite old bug.
I think the check here should be at `subdirs[meancount]` instead of `subdirs[i]`.

At the same time I am not sure whether it is a good idea to have the array size of `dirsx`, `dirsy` and `subdirs` hardcoded to 64 while the max of `meancount` is defined indirectly by `SEARCH_SIZE` and `NUM_BLOCKS.` If I remember right, the assumption is that it should not exceed 64, so it might be fine with the current hardware but it is not nice since it does not adapt correctly.

leonkaihao, 2017-12-21:
Yes, that's true. We also can find the hard code in micro ‘ABSDIFF’. It seems that a 64x64 image processing is not a option but a necessity.

mike239x, 2020-02-05:
+1, that should have been `subdirs[meancount]` aka `mindir`.
Also +1 regarding `ABSDIFF`, though I think one should open another issue for that one. Also, in the code there is the `compute_sad_8x8` which does exactly the same and does not have restriction of width of 64px, but that function is unused and  `ABSDIFF` is used instead...

commit 9ebe6463 (2020-02-11): Fix #98
commit d2875c4e (2020-02-11): Merge pull request #124 from mike239x/pr

Fix #98 second try
closed 2020-02-11 by julianoes