PX4/PX4-Flow #101: PX4FLOW publishing zeros to DISTANCE_SENSOR.current_distance

rr2-ip, 2018-08-29:
Hi,
I am trying to get PX4FLOW working on my Intel Aero RTF Drone, using the PX4FLOW board and a LeddarOne rangefinder. I have the LeddarOne **and** the PX4FLOW board plugged into the Telemetry port of the drone. I do not have an ultrasonic module attached to the PX4FLOW board.
The telemetry port has 6 pins - VCC, GND, TX, RX, SCL, SDA. I'm using the TX and RX pins for the rangefinder, and the SCL, SDA pins for the PX4FLOW. I know this is possible as seen here: https://github.com/intel-aero/meta-intel-aero/issues/315

My issue is that both devices are publishing distance data to DISTANCE_SENSOR.current_distance. The PX4FLOW only publishes zero values, which means that the accurate rangefinder readings are interwoven with zeros. Please see the attached picture for an idea of what I mean. 
[image]

The PX4FLOW publishes DISTANCE_SENSOR data **even when there are no other sensors enabled or attached to the drone**.

How can I stop the PX4FLOW publishing DISTANCE_SENSOR data, but still publish OPTICAL_FLOW_RAD data?

Thanks in advance

ChristophTobler, 2018-08-29:
I don't think that this is possible as there is no parameter for that. Maybe we should create one.?
You could however remove https://github.com/PX4/Firmware/blob/master/src/drivers/px4flow/px4flow.cpp#L591-L604 in the Firmware and flash that version.

rr2-ip, 2018-08-29:
@Torgenn Did you encounter this issue when you got LeddarOne + PX4FLOW working on your Intel Aero RTF drone?

rr2-ip, 2018-08-29:
@ChristophTobler I'll give that a try and report back. Thanks for the very quick response!

ChristophTobler, 2018-08-29:
I never had a PX4Flow attached to an Intel Aero as it already has a downward facing camera and an optical flow app.

rr2-ip, 2018-08-29:
That didn't work for me. Take a look at the issue I raised here: https://github.com/intel-aero/meta-intel-aero/issues/372

The FPS of the downward facing camera is too low. I think it's a fault with our drone's camera.

So the PX4FLOW camera is a work around because of that issue.

Torgenn, 2018-08-29:
@rr2-ip I never had that issue, the distance_sensor.current_distance was showing the leddar1 values only

ChristophTobler, 2018-08-30:
>I never had that issue, the distance_sensor.current_distance was showing the leddar1 values only

On other platform I've never had this issue as well. If there are two distance sensors, they will be published on different uorb instances and e.g. EKF2 will only use one of them.
I would need to check if mavlink uses several instances and sends a mix of them.

rr2-ip, 2018-08-30:
Both sensors are plugged into the one Telemetry port, and both show up as ID = 0. Could this be the reason that the distance readings are getting mixed up?

rr2-ip, 2018-08-30:
@ChristophTobler When I comment out the lines you suggested and flash my drone, the px4flow driver is absent when I check for it in MAVLINK CONSOLE. Before I comment out that section, the driver shows up no problem.

rr2-ip, 2018-08-30:
@Torgenn Thanks for the response!

ChristophTobler, 2018-08-30:
>When I comment out the lines you suggested and flash my drone, the px4flow driver is absent when I check for it in MAVLINK CONSOLE. Before I comment out that section, the driver shows up no problem.

What do you mean with the driver shows up in the mavlink console? In QGC Analyze? Do you mean you don't see a distance sensor? That would mean that your LeddarOne isn't working.
If you comment out the lines only the distance sensor from the PX4Flow module should not be pubished. Optical flow should still be published. Can you verify that?

rr2-ip, 2018-08-30:
In QGC, there is a MAVLINK console that is basically a terminal to access the drone's flight controller and associated commands (like starting or stopping a sensor driver). 

When I comment out the lines as you suggested, OPTICAL_FLOW_RAD is **not visible**, but DISTANCE_SENSOR is **visible**, displaying LeddarOne data only (no zeros weaved in between). px4flow driver is **not visible** in MAVLINK console (appearing here means that the driver has been built for the specific sensor and is ready to use).

When they are not commented out, both OPTICAL_FLOW_RAD and DISTANCE_SENSOR  are **visible** in MAVLINK Inspector in QGC. The only problem here is that the LeddarOne readings are corrupted by zeros from the PX4FLOW (See original picture). px4flow driver is **visible** in MAVLINK Console.

ChristophTobler, 2018-08-30:
Ah ok, that's weird. Can you check if PX4Flow is running? `px4flow status` in that console
Are you using a "self compiled" Firmware in both cases? (not a release for example) And did you make sure you only commented the distance sensor publication and not both?

rr2-ip, 2018-08-30:
`px4flow status` returns `px4flow: command not found` if the px4flow driver is not visible in the MAVLINK console, otherwise it returns some test data that makes it seem like it is working. 

I'm not sure what you mean by self compiled, but I have forked the PX4/Firmware repository and made the changes there. Then I am running from the Home directory on my Ubuntu PC:
`git clone https://github.com/rr2-ip/Firmware/`
`cd ~/Firmware`
`make aerofc-v1_default`
Next I copy the `aerofc-v1_default.px4` file located in `~/Firmware/build/nuttx_aerofc-v1_default/` onto my drone, then running (from my drone)
`sudo aerofc-update.sh aerofc-v1_default.px4`

It's also worth mentioning:

1. that the PX4FLOW has power (LEDs are on)
2. once I revert the changes to the original code (as you would find in PX4/Firmware/), rebuild and reflash the image, the px4flow is still not present in MAVLINK console (it was before). OPTICAL_FLOW_RAD is **not visible**, but DISTANCE_SENSOR is **visible**.

ChristophTobler, 2018-08-30:
That's what I thought. In your original test you didn't use a self compiled Firmware binary. You can't use the driver because it has not been compiled for the AeroFC target. You need to add the PX4Flow driver to https://github.com/PX4/Firmware/blob/master/cmake/configs/nuttx_aerofc-v1_default.cmake#L6 (`drivers/px4flow`) to that list.

rr2-ip, 2018-08-30:
That's confusing. Why was the PX4FLOW driver present when I clone directly from the PX4/Firmware github?

ChristophTobler, 2018-08-30:
>Why was the PX4FLOW driver present when I clone directly from the PX4/Firmware github?

That doesn't change anything. Have you actually compiled and uploaded the Firmware yourself before? Could be that the previous version on your Aero was an Intel version that had the driver added to that target.
Anyways, is it now working for you?

rr2-ip, 2018-08-30:
I've done it a few times but I don't know some of the intricacies of building custom firmware. The first time I did it after forking, the px4flow driver was visible and working... strange. I'll be sure to take special note of the nuttx.cmake file from here on out.

PX4FLOW is still somehow publishing zeros to DISTANCE_SENSOR, even after commenting out the code you suggested. I have even tried disabling and unplugging all other sensors and I still get DISTANCE_SENSOR returning zeros at approx 10Hz.

ChristophTobler, 2018-08-30:
Hm... Can you try commenting all occurrences of `_distance_sensor_topic` in the driver and disable/unplug the LeddarOne?

rr2-ip, 2018-08-31:
Uncommenting all occurrences of _distance_sensor_topic fixed the issue for me. Thank you very much for your assistance.

closed 2018-08-31 by rr2-ip

ChristophTobler, 2018-08-31:
Glad to help :) Thanks