PX4/PX4-Flow #109: Why use flow as a angular sensor rather than a linear velocity/position sensor?

MerlotQ, 2019-02-28:
I see px4 firmware use flow rad msg to fuse with imu. Why not use it as a position or veloocity compensator? I test the px4flow recently and found that the linear velocities flow msg gives seems good to use.

closed 2019-03-06 by MerlotQ

mohammedari, 2019-03-11:
I think distance from the sensor to ground is required for conversion from optical flow to linear velocity, but this conversion is have to be done in PX4 side as PX4 supports use px4flow with various distance sensor other than sonic sensor on px4flow such as lidar sensors.

So, the answer for the question is because px4flow itself can not convert it. However, I am not familiar with this conversion, so I am also wondering why rad value is used for the message.

MerlotQ, 2019-03-11:
@mohammedari Yes. The flow sensor cannot give absolute velocity. It needs a scale. So the integrated angular velocity namely in 'rad' from is in use. The value devided by integrated timespan gives angular velocity. And then devided again by distance to ground returns linear velocity in a metric scale. I think this is the reason to use this delta angle to represnet optical flow rate(equivalent of LOS rate). See [this explanation](https://dh8.kr/workshop/small_uas_state_estimation/CourseNotes.pdf).

mohammedari, 2019-03-12:
@MerlotQ Thank you for the link. I have now understand why flow_y is with 'rad' form as `V_y / Z` is `tan(\theta)` and this approximately equals to `\theta` while it is small. Thanks!