PX4/PX4-Flow #113: px4flow sonar no distance. Help!

pxfdale, 2019-06-19:
I followed the steps( I have flashed px4flow.px4 & px4flow-klt-06Dec2014.px4... ) in the Internet to use px4flow. ( [http://ardupilot.org/copter/docs/common-px4flow-overview.html](url) )

px4flow works but sonar seeems to not work. flow_x and flow_y works well, but distance is always 0.

Guys, what's the steps you use sonar? Did I miss any steps? Is the output of sonar "distance" or others?

( Besides, I used QGroundControl, pixhawk2.4.8, px4fmu_v3-default.px4, px41.8.2 )

This is the pictures when flow camera is linked to PC.

[image]
[image]
[image]
[image]

I have worked for it for a long time but still do not know where the problem is. Hoping to get help from you! Thank you!

mike239x, 2020-02-05:
Do you have a sonar attached to px4flow? If not then the behavior you observe is to be expected.

rk800506, 2020-03-05:
you must be using a distance sensor along with px4flow

Chick92, 2020-06-19:
I am having the exact same issue, have you made any progress?

Does the jumper for TTL on the sonar have to be soldered? It could possibly be a broken sonar unit, but i don't know if the PX4FLOW uses the analog, pwm, or serial (in TTL or RS232) mode? Some clarification on this will be helpful, as i can then being to debug. 

Is the sonar still supported on the PX4FLOW? I'm not intending to use it with a flight controller, i'm using it for a robotics application and require the absolute position change. I could just integrate my own laser range finder and scale accordingly, but i really don't want to have to do more work than required. 

Ben

Chick92, 2020-06-20:
For anyone else reading this - 

Solder the jumper pad on the ultrasonic sensor to enable a ttl serial output, and solder the 5v jumper on the PX4FLOW, the ultrasonic will then work and give data. 

It's genuinely ridiculous that i had to trawl obscure forums to do with rovers to find that out. 
It's not listed in any of the documentation i could find for the PX4FLOW sensor.

rk800506, 2020-06-20:
I am not using sonar. I have used a TFMini LIDAR as a range finder.