PX4/PX4-Flow #114: Vehicle reject position control when using optical flow

MKargus0, 2019-07-13:
My px4(1.9.2) Quad reject position control when using Px4 flow sensor(Holybro) with PX4 Pro latest firmware
I use i2c bus for conection.

SENS_FLOW_ROT = 270
I try to use LPE 
LPE_FUSION = 4
LPE_FLW_QMIN  = 10.
LPE_FLW_SCALE = 1.0.
LPE_FLW_R = 0.1.
LPE_FLW_RR = 0.0.
and EKF 2 estimator 
EKF2_AID_MASK =  2
EKF2_OF_DELAY = 0.
EKF2_OF_QMIN  = 10.
EKF2_OF_N_MIN = 0.05.
EKF2_OF_N_MAX = 0.2

Can you help me with this ? Thank you.

.

kornat, 2019-10-01:
I have exactly the same problem.

alexcherpi, 2019-12-27:
I also have the same issue.

zhanghaijason, 2020-07-02:
Hi, is there any progress in using optical flow sensor? I am running the same rejection position problem with a flow breakboard. Thanks