PX4/PX4-Flow #115: Firmware version onboard

darkom-ltu, 2019-07-25:
Hello,

Is it possible to read which firmware version is currently on PX4FLOW?
I notice performance degradation when I upload the newest firmware with exactly the same parameters I had on the one which sensor came with.

I also wonder if it would be possible to get 'hex' out of the PX4FLOW sensor out as a 'backup' and flash another sensor with it? I currently have 1 working on older firmware which works well and 2 on the newest one with much worse low light performance, even all parameters are the same.

mike239x, 2020-02-05:
Can you check that EXPOSURE_MAX and BRIGHT params are the same as before?

tstellanova, 2020-07-18:
I was wondering about the version currently available via QGroundControl, so I grabbed it from AWS and ran:

```
cat px4flow.px4 | strings
```
This gave the following info: 
```
  "git_identity": "1bb4e50-dirty", 
```
So apparently that build is from [code submitted August 2017[(https://github.com/PX4/Flow/commit/1bb4e50d305ebd952851e8cbea4b3312e9744360).