PX4/PX4-Flow #116: Binning

galdalali, 2019-08-12:
Can someone explain the idea behind the binning operation?

Thanks, Gal.

```
const float focal_length_px = ((global_data.param[PARAM_FOCAL_LENGTH_MM])
			/ (4.0f * 6.0f) * 1000.0f); //original focal lenght: 12mm pixelsize: 6um, binning 4 enabled
```

mike239x, 2020-02-14:
The camera (MT9V034) has standard output of 752x480, but we use 4x4 binning, so the images have size of 188x120. So a single pixel on the resulting image has size of 24μm instead of original 6μm.

timcamber, 2020-04-16:
In the paper describing the PX4 (Honegger, "An Open Source and Open Hardware Enabled Embedded Metric ..."), it says "When pixel binning is enabled...", implying binding is a configurable parameter. Based on the code repository, it looks like this is fixed at binning 4. Which is correct? Thanks!

tpetri, 2020-04-17:
The paper references the Aptina MT9V034 sensor at that point in the text, the context being: Natively the sensor allows 60Hz at 752Hx480V, but if you activate binning [on the sensor] it can go upto 250Hz at 188Hx120V.
So yes, it is a configurable parameter of the hardware sensor itself and this configuration is done in the code, it is set to 4.
Implying from the paper that it is a user configurable setting over the params interface goes a bit far, if you hoped for that ;)

timcamber, 2020-04-20:
Thanks! This is very helpful. I appreciate it.