PX4/PX4-Flow #128: Error in build

rushilkhurana, 2020-07-08:
I cloned the directory, and followed the instructions. I was able to successfully do make archives. However, when I run 'make', I get the following error: 

`In file included from /Users/sylar/PX4/Flow/src/lib/uavcan/libuavcan/include/uavcan/node/node.hpp:14:0,
                 from /Users/sylar/PX4/Flow/src/lib/uavcan/libuavcan/include/uavcan/uavcan.hpp:14,
                 from /Users/sylar/PX4/Flow/src/lib/uavcan/libuavcan_drivers/stm32/driver/include/uavcan_stm32/uavcan_stm32.hpp:7,
                 from /Users/sylar/PX4/Flow/src/modules/uavcannode/uavcannode_main.hpp:37,
                 from /Users/sylar/PX4/Flow/src/modules/uavcannode/uavcannode_main.cpp:39:
/Users/sylar/PX4/Flow/src/lib/uavcan/libuavcan/include/uavcan/protocol/node_status_provider.hpp:14:10: f**atal error: uavcan/protocol/NodeStatus.hpp: No such file or directory**
 #include <uavcan/protocol/NodeStatus.hpp>
          ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
compilation terminated.
make[2]: *** [uavcannode_main.cpp.o] Error 1
make[1]: *** [/Users/sylar/PX4/Flow/Build/px4flow-v2_default.build//Users/sylar/PX4/Flow/src/modules/uavcannode/module.pre.o] Error 2
make: *** [/Users/sylar/PX4/Flow/Build/px4flow-v2_default.build/firmware.px4] Error 2`

Any ideas on how to fix this?

RealityVoid, 2020-07-08:
Hmm, I'm not too familiar with this build system but I'm going to take a crack at this. 

 - On my machine (popos 20.04) it builds just fine so I think the repo is fine, it's _prooobably_ something with your setup. 
- I see NodeStatus.hpp on my end inside ./Build/px4flow-v2_default.build/home/vladr/git/Flow/src/modules/uavcannode/dsdlc_generated/uavcan/protocol/NodeStatus.hpp - might want to check if you have such a file in your system. It seems the file is generated during the make process. So it's possible some dependancy is not installed properly and the file is not being generated.
 - the uavcan is its own repo that gets cloned  during build. Do you have it succesfully cloned in your system? What is the answer of git status? Do you have it in ~/Flow/src/lib/uavcan ?

rushilkhurana, 2020-07-08:
I have a working directory until- 

`/Users/sylar/PX4/Flow/Build/px4flow-v2_default.build/Users/sylar/PX4/Flow/src/modules/uavcannode/dsdlc_generated/uavcan`

It has only two files: 

> CoarseOrientation.hpp	Timestamp.hpp

It does not have any folder for protocol which should contain the missing file. 

Yes, the directory: 

> ~/Flow/src/lib/uavcan 

exists, and it has a bunch of files.

When I do a git status in this folder, I get: 

> HEAD detached at 4dff5c22
> nothing to commit, working tree clean

So, it looks like it did successfully clone.

rushilkhurana, 2020-07-18:
I am not sure why it solved the issue, but for future reference- on my setup, I had to download the [PX4 Firmware](https://github.com/PX4/Firmware) and follow the Clang instructions here: https://dev.px4.io/v1.9.0/en/setup/dev_env_linux_ubuntu.html#clang

In the Clang instructions, I removed the rpi specific paths. 

When I built and uploaded the Flow repository afterwards, I did not get the above issue and was able to make all. 

Even though this is not a requirement listed for Mac, this is the only way I could get it to successfully work with my setup. 

I am having other (possibly related?) issues with the output, and I will open a new issue for that.

closed 2020-07-18 by rushilkhurana