PX4/PX4-Flow #130: <uavcan/protocol/GetNodeInfo.hpp> not detected

akvavit01, 2020-10-15:
When building using make all, GetNodeInfo,hpp is not found

[image]

RPFey, 2021-01-04:
Hi, I encounter a similar problem. I can't find file `NodeStatus.hpp`. I find the same module `libuavcan`  in PX4 repo Firmware and I can build it. So I believe the building environment is wrong.
I check the build output and find the python generator fail. I disable the conda environment and use python 2.7 to build, and it pass.
I think you should check the building environment according to PX4 Guide and use the same environment to build PX4-FLOW.
Good Luck!

sbaccam90, 2021-05-13:
@akvavit01 and @RPFey  I think I solved it. The problem happens with the new versions of Python; I am using the 3.8.5 which comes with Ubuntu 20.04, [here more information](https://stackoverflow.com/questions/51700960/runtimeerror-generator-raised-stopiteration-every-time-i-try-to-run-app). Specifically, you need to handle the stopIteration exception version in **/PX4-Flow/src/lib/uavcan/libuavcan/dsdl_compiler/libuavcan_dsdl_compiler/__init__.py** since after Python 3.6 this will end up in an error. In this case, the error stops the generation process of some files such as Nodeinfo.hpp. Specifically, I modified the line 300 in _init.py_ for handling the exception:
`            try:
                last = next(it)
            except StopIteration:
                return`