PX4/PX4-Flow #132: Low and fluctuating quality of flow over i2c while USB works fine

ashwinlele, 2021-05-09:
I am trying to use px4flow optical flow with Lidar Lite v3 for optical flow based EKF2 position stabilization on NXP FMUK66. My optical flow camera gives high quality output (>200) when connected over USB. The images are of good quality and I tested it over multiple surfaces and illumination.

However, the quality drops suddenly when connected to the flight controller over i2c and fluctuates heavily. I initially though that this might be because of the noisy sonar measurements. However, even with highly accurate lidar distance sensor, the quality has remained low. The typical output quality I see is shown in this image --> https://drive.google.com/file/d/1429njfhc_QoDVoVeUGtqUDGj6C9hNeyy/view?usp=sharing

My flow camera is powered over i2c while only SDA,SCL and GND are connected to the flight controller to ensure sufficient power delivery. Can someone please help me out?

Thanks in advance!

davids5, 2021-05-10:
chac the speed of the I2C it 100 kHz or 400 kHz? What voltages is connected to the Flow?

ashwinlele, 2021-05-10:
Thank you for the reply. I am starting the sensor with the command px4flow start -X -f 400 to start it with 400 kHz. The file "PX4-Flow-master\src\modules\flow\i2c.c" also defines: i2c_init.I2C_ClockSpeed = 400000; 

However, I am not sure if this is getting accepted as I see the rate at 10 Hz in MAVlink inspector as shown in the image: https://drive.google.com/file/d/1jgpv6AhtgZqyEU2CQNIccZ3cIc2ibJKS/view?usp=sharing
The output of my i2cdetect command in MAVLink console is: https://drive.google.com/file/d/1jQ7DPgm6bnhJ7smz4Gac1v7P4oXQ5tqf/view?usp=sharing

Is there any place that I should be modifying to check my speed? The voltage is 5V USB which powers the PX4flow camera from my laptop. px4flow is connected to FMUK66 over i2c with SDA SCL and GND connected.

Thanks in advance!