PX4/PX4-Flow #134: Live video view Totally Striped with VIDEO_ONLY  = 0

rohitk1979, 2021-09-13:
Hello. I have CUAV PX4Flow version 2.3 board:

[image]
The firmware is v1

Running QGroundControl, I am able to view the live video feed. However, it is totally jumbled up with horizontal stripes:

[image]

When I set VIDEO_ONLY=1, i.e. in calibration mode, I am able to see a crisp and clear image:

[image]

In normal use, with VIDEO_ONLY=0, the optical flow readings are naturally all over the place for to unclear and striped image.

Any help will be appreciated as this is a pretty expensive module. Don't want it get bricked...

Thanks.

mike239x, 2021-09-17:
I'd assume something is wrong with image width/height, which would cause image wrapping and produce strips just like what you see. I can't tell exactly where the problem occurs though...

rohitk1979, 2021-09-18:
Could you point me where in the code can i tinker to see if there any effect on this?
Or could it be a hardware fault? Do I need to replace a resistor, regulator etc.?

rohitk1979, 2021-10-06:
I tried every available firmware with this module with version 2.3 inscribed. Even tried to build from source.
I even contacted engineers at CUAV, who have me a firmware which I also tried but it did not work.

However, I purchased a new module with version 1.3 clearly marked, and it worked out of the box perfectly the first time- clearly showing the camera view.

How could i approach "repairing" the v2.3 module?